robsoncouto/arduino-songs #1: Pac Man theme

adriens, 2019-09-01:
Hi, I love your repo, it trully rocks ! Do you intend to implement PacMan theme ?

I'll use your work in a fun DIY project, I'l quote you.

Have a nice day.
Kind Regards,
Adrien

adriens, 2019-09-01:
I found this : https://musescore.com/user/85429/scores/107109

adriens, 2019-09-01:
https://www.youtube.com/watch?v=x6ft1POjDjM

adriens, 2019-09-01:
I'm starting to implement it. I'll make you a PR if you're ok with that 👍

adriens, 2019-09-01:
First notes....but not very satisfied for now :

 ```
 NOTE_B5,6,
  NOTE_B6,6,
  NOTE_FS6,6,
  NOTE_DS6,6,

  NOTE_B6,6,
  NOTE_F6,6,
  NOTE_E6,6,

  NOTE_C5,6,
  NOTE_C6,6,
  NOTE_G6,6,
  NOTE_F5,6,

  NOTE_C6,6,
  NOTE_G6,6,
  NOTE_E6,6
```

robsoncouto, 2019-09-02:
Hi Adrien!
I'm glad :)

You got some of the notes right, but the timing and octave is also important. 
Please see if sounds alright, I don't have an Arduino board with me at the moment to test. I have only used the treble notes.
```
NOTE_B4,16,
NOTE_B5,16,
NOTE_FS5,16,
NOTE_DS5,16,

NOTE_B5, 32,
NOTE_FS5,-16,
NOTE_DS5,8,

NOTE_C5,16,
NOTE_C6,16,
NOTE_G6,16,
NOTE_E6,16,

NOTE_C6,32,
NOTE_G6,-16,
NOTE_E6,8,

NOTE_B4,16,
NOTE_B5,16,
NOTE_FS5,16,
NOTE_DS5,16,

NOTE_B5, 32,
NOTE_FS5,-16,
NOTE_DS5,8,

NOTE_DS5, 32,
NOTE_E5, 32,
NOTE_F5, 32,

NOTE_F5, 32,
NOTE_FS5, 32,
NOTE_G5, 32,

NOTE_G5, 32,
NOTE_GS5, 32,
NOTE_A5, 16,
NOTE_B5, 8
```

robsoncouto, 2019-09-02:
Sorry, i have deleted my previous response thinking it was input twice. But you probably will receive it as email.
Anyway, thanks for your words and please test the following code, I don't have an Arduino board with me at the moment :)

```
NOTE_B4,16,
NOTE_B5,16,
NOTE_FS5,16,
NOTE_DS5,16,

NOTE_B5, 32,
NOTE_FS5,-16,
NOTE_DS5,8,

NOTE_C5,16,
NOTE_C6,16,
NOTE_G6,16,
NOTE_E6,16,

NOTE_C6,32,
NOTE_G6,-16,
NOTE_E6,8,

NOTE_B4,16,
NOTE_B5,16,
NOTE_FS5,16,
NOTE_DS5,16,

NOTE_B5, 32,
NOTE_FS5,-16,
NOTE_DS5,8,

NOTE_DS5, 32,
NOTE_E5, 32,
NOTE_F5, 32,

NOTE_F5, 32,
NOTE_FS5, 32,
NOTE_G5, 32,

NOTE_G5, 32,
NOTE_GS5, 32,
NOTE_A5, 16,
NOTE_B5, 8
```

adriens, 2019-09-02:
Hey, thanks a lot for your ticket update 🎆 , here is the output result :

https://photos.app.goo.gl/4jH7zkT6w3haw13z5

Sound pretty good, and the tempo is amazing fast ! 😍

adriens, 2019-09-02:
Do you want me to make you a PR with your own code ?

adriens, 2019-09-02:
Includeing the README, etc ?

robsoncouto, 2019-09-02:
I searched online and it seems the correct tempo is 105bpm. 
Thanx. I can update the repo later, don't worry. :)

adriens, 2019-09-02:
👍  I'm not worrying, I just find your repo very cool !

adriens, 2019-09-03:
Do you have any Twitter or linkedIn page 🙏  ❔

robsoncouto, 2019-09-03:
> 👍 I'm not worrying, I just find your repo very cool !

I'm glad it is helpful for you :)

>Do you have any Twitter or linkedIn page 🙏 ❔

Unfortunately I don't have either

adriens, 2019-09-04:
So I'll point to your Github project for acknowledgments 👍

adriens, 2019-09-11:
Hey, here are some news : http://bit.ly/2mawDzR

Sorry I even don't know your real name, but collaborating with you was a reallllllly cool experience !

robsoncouto, 2019-09-19:
I have uploaded the pacman song in 1eb90e101b2981371f47826ccd158d2a70dfdef6.
Thanks for your suggestion, Adrien. 

Best wishes
Robson

closed 2019-09-19 by robsoncouto