robsoncouto/arduino-songs #5: New_song_generation

SaiSandeepKantareddy, 2020-07-12:
HI, I really like your repo, can you provide further details about how can we start doing it on our own for any other songs? that will be really helpful to everyone.

robsoncouto, 2020-07-14:
Hello @SaiSandeepKantareddy 

There´s really no easy, quick or automatic way of doing this. 
First, one needs a basic understanding of how to read music sheet. This [video ](https://www.youtube.com/watch?v=ZN41d7Txcq0)is a great start. 
Then they need to find a score of the song, specially for flute or other instrument that can play only a single note at a time. I usually find these at https://musescore.com/

With libraries you can also play more than a note at a time, although this depends on the board used, so I stick to monophonic melodies in order for the code here to run on every board.

robsoncouto, 2020-07-22:
Since there was no further response, I will close this.

closed 2020-07-22 by robsoncouto