robsoncouto/arduino-songs #10: Esp32?

Snooze4, 2021-01-06:
Hello, can those songs be used on esp32 somehow?

robsoncouto, 2021-01-07:
Hi @Snooze4 , if you are using Arduino to program the esp32, there should not be a problem.
It's highly unlikely that it is a library problem, unless you installed some library that changed the default behaviour of tone().
Anyways, you should not need any library to run this code. It only depends on standard arduino functions.

Please check that you are using the correct pin and also that the pin you are using can be used for this purpose. You can change the pin number in the code, please have a look.

Also, I don't have an Esp32, so i can not confirm it works. I don't see why it should not though.

Snooze4, 2021-01-08:
Well esp32 does not understand tone.

closed 2021-01-10 by robsoncouto

umsmin, 2022-11-11:
Sorry for my late help.
But I think it will maybe help others.

It works on ESP32 with a little library and modifications.

[(https://github.com/lbernstone/Tone32)](https://github.com/lbernstone/Tone32)

In the example you see how to use it, but for now.

Add this on top of the code:

```
// The tone library for ESP32
#include <Tone32.h>

// we use the PWM channel 0 for here
int buzzerChannel = 0;
```

Change this:

`tone(buzzer, pgm_read_word_near(melody+thisNote), noteDuration * 0.9);`

To this:

`tone(buzzer, pgm_read_word_near(melody+thisNote), noteDuration * 0.9, buzzerChannel);`

From this:

```
} else if (divider < 0) {
      // dotted notes are represented with negative durations!!
      noteDuration = (wholenote) / abs(divider);
      noteDuration *= 1.5; // increases the duration in half for dotted notes
}
```

To this
```
else if (divider < 0) {
      // dotted notes are represented with negative durations!!
      noteDuration = (wholenote) / abs(divider);
      noteDuration = noteDuration * 1.5; // increases the duration in half for dotted notes
}
```

And also this:

`noTone(buzzer);`

To this:

`noTone(buzzer, buzzerChannel);`

That's it.