robsoncouto/arduino-songs #27: Buttons don't work while buzzer is playing

PedroVillasBoas, 2022-11-10:
Hi!

Firstly, I loved the songs. And thank you for your work!

Secondly, I'm building a little game with an arduino mega 2560. It's really simple. 2 People will battle to see who has the faster finger press.

Basically, I have 5 buttons (2 arcade and 3 simple push buttons), 1 LCD i2C 16,2 screen, 1 4Digit 7 segment display and a buzzer.

On my code (I haven't finished it yet) I was trying to test the functionalities (pressing any of the buttons while the buzzer is playing) I implemented so far, and the buttons stopped working while the music was playing.

If I comment the function that makes the music play and re-upload to my arduino, all of the buttons work again.

I don't know why this is happening.

If anyone could help me, I would be really grateful.

Anyway, thank you!

(I'll post the code here!)
 (I've divided the code into 2 tabs. The main one has almost everything. The second one just has the Music() function).
    (FYI, I'm brazilian. So some parts of the code has a comment or two in portuguese. :/ ).

//////////////////
//CODE ONE//
//////////////////

`// Libraries
#include <Wire.h>
#include <LiquidCrystal_I2C.h>

// Define cada nota para um frequencia em hertz
#define NOTE_B0  31
#define NOTE_C1  33
#define NOTE_CS1 35
#define NOTE_D1  37
#define NOTE_DS1 39
#define NOTE_E1  41
#define NOTE_F1  44
#define NOTE_FS1 46
#define NOTE_G1  49
#define NOTE_GS1 52
#define NOTE_A1  55
#define NOTE_AS1 58
#define NOTE_B1  62
#define NOTE_C2  65
#define NOTE_CS2 69
#define NOTE_D2  73
#define NOTE_DS2 78
#define NOTE_E2  82
#define NOTE_F2  87
#define NOTE_FS2 93
#define NOTE_G2  98
#define NOTE_GS2 104
#define NOTE_A2  110
#define NOTE_AS2 117
#define NOTE_B2  123
#define NOTE_C3  131
#define NOTE_CS3 139
#define NOTE_D3  147
#define NOTE_DS3 156
#define NOTE_E3  165
#define NOTE_F3  175
#define NOTE_FS3 185
#define NOTE_G3  196
#define NOTE_GS3 208
#define NOTE_A3  220
#define NOTE_AS3 233
#define NOTE_B3  247
#define NOTE_C4  262
#define NOTE_CS4 277
#define NOTE_D4  294
#define NOTE_DS4 311
#define NOTE_E4  330
#define NOTE_F4  349
#define NOTE_FS4 370
#define NOTE_G4  392
#define NOTE_GS4 415
#define NOTE_A4  440
#define NOTE_AS4 466
#define NOTE_B4  494
#define NOTE_C5  523
#define NOTE_CS5 554
#define NOTE_D5  587
#define NOTE_DS5 622
#define NOTE_E5  659
#define NOTE_F5  698
#define NOTE_FS5 740
#define NOTE_G5  784
#define NOTE_GS5 831
#define NOTE_A5  880
#define NOTE_AS5 932
#define NOTE_B5  988
#define NOTE_C6  1047
#define NOTE_CS6 1109
#define NOTE_D6  1175
#define NOTE_DS6 1245
#define NOTE_E6  1319
#define NOTE_F6  1397
#define NOTE_FS6 1480
#define NOTE_G6  1568
#define NOTE_GS6 1661
#define NOTE_A6  1760
#define NOTE_AS6 1865
#define NOTE_B6  1976
#define NOTE_C7  2093
#define NOTE_CS7 2217
#define NOTE_D7  2349
#define NOTE_DS7 2489
#define NOTE_E7  2637
#define NOTE_F7  2794
#define NOTE_FS7 2960
#define NOTE_G7  3136
#define NOTE_GS7 3322
#define NOTE_A7  3520
#define NOTE_AS7 3729
#define NOTE_B7  3951
#define NOTE_C8  4186
#define NOTE_CS8 4435
#define NOTE_D8  4699
#define NOTE_DS8 4978
#define REST      0

// change this to make the song slower or faster
int tempo = 108; // 108 is standard.

// change this to whichever pin you want to use
int buzzer = 27;

// notes of the moledy followed by the duration.
// a 4 means a quarter note, 8 an eighteenth , 16 sixteenth, so on
// !!negative numbers are used to represent dotted notes,
// so -4 means a dotted quarter note, that is, a quarter plus an eighteenth!!
int melody[] = {
  
  NOTE_E4,2, NOTE_G4,4,
  NOTE_D4,2, NOTE_C4,8, NOTE_D4,8, 
  NOTE_E4,2, NOTE_G4,4,
  NOTE_D4,-2,
  NOTE_E4,2, NOTE_G4,4,
  NOTE_D5,2, NOTE_C5,4,
  NOTE_G4,2, NOTE_F4,8, NOTE_E4,8, 
  NOTE_D4,-2,
  NOTE_E4,2, NOTE_G4,4,
  NOTE_D4,2, NOTE_C4,8, NOTE_D4,8, 
  NOTE_E4,2, NOTE_G4,4,
  NOTE_D4,-2,
  NOTE_E4,2, NOTE_G4,4,

  NOTE_D5,2, NOTE_C5,4,
  NOTE_G4,2, NOTE_F4,8, NOTE_E4,8, 
  NOTE_F4,8, NOTE_E4,8, NOTE_C4,2,
  NOTE_F4,2, NOTE_E4,8, NOTE_D4,8, 
  NOTE_E4,8, NOTE_D4,8, NOTE_A3,2,
  NOTE_G4,2, NOTE_F4,8, NOTE_E4,8, 
  NOTE_F4,8, NOTE_E4,8, NOTE_C4,4, NOTE_F4,4,
  NOTE_C5,-2, 
  
};

// sizeof gives the number of bytes, each int value is composed of two bytes (16 bits)
// there are two values per note (pitch and duration), so for each note there are four bytes
int notes = sizeof(melody) / sizeof(melody[0]) / 2;

// this calculates the duration of a whole note in ms
int wholenote = (60000 * 4) / tempo;

int divider = 0, noteDuration = 0;

// Seta o adress da LCD para 0x3F para 16 chars e 2 linhas do display
LiquidCrystal_I2C lcd(0x3F, 16, 2);

// Definindo as variaveis
  // Botoes
int numberButtons = 5;   // Numero total de botoes
int startPin = 22;       // Onde comeca a porta dos botoes
int buttons[5];          // Lista com todos os botoes
boolean buttonsBool[5];  // Bool pra checar o estado do botao

  // Buzzer
const int miniSom = 27;  // Pin do MiniSom
  // Jogo
int pressNumber = 0;  // Numero total de vezes que vao ser apertadas no jogo

void setup() {
  Serial.begin(9600);
  // Inicilizando todos os botoes em seus respectivos pins
  for (int i = 0; i < numberButtons; i++)
    buttons[i] = i + startPin;
  // Setando todos os botoes para o estado de falso
  for (int i = 0; i < numberButtons; i++)
    buttonsBool[i] = false;
  // Setando todos os botoes para o estado de input pullup
  for (int i = 0; i < numberButtons; i++)
    pinMode(buttons[i], INPUT_PULLUP);

  pinMode(miniSom, OUTPUT);

  // inicializando o lcd
  lcd.init();
  // Liga a luz de fundo e limpa a tela com o clear
  lcd.backlight();
  lcd.clear();
}

void loop() {
  //Music();
  ButtonPressing();
  /*
  digitalWrite(miniSom, HIGH);
  delay(1000);
  digitalWrite(miniSom, LOW);
  delay(1000);
  */
}

void ButtonPressing() {
  // while (digitalRead(buttons[2] == HIGH))
  // {
  for (int i = 0; i < numberButtons; i++) {
    if (!buttonsBool[i]) {
      if (digitalRead(buttons[i]) == LOW) {
        if (i == 0) {
          pressNumber++;
          lcd.clear();
          lcd.print(pressNumber);
          buttonsBool[i] = true;
        } else if (i == 1) {
          pressNumber += 10;
          lcd.clear();
          lcd.print(pressNumber);
          buttonsBool[i] = true;
        } else if (i == 2) {
          lcd.clear();
          pressNumber = 0;
        } else if (i == 3) {
          lcd.clear();
          lcd.print("Player 1 funcionando!");
          buttonsBool[i] = true;
        } else if (i == 4) {
          lcd.clear();
          lcd.print("Player 2 funcionando!");
          buttonsBool[i] = true;
        }
      }
    } else {
      if (digitalRead(buttons[i]) == HIGH) {
        buttonsBool[i] = false;
      }
    }
  }
  delay(15);
}`

//////////////////
//CODE TWO//
//////////////////
`void Music(){
    // iterate over the notes of the melody. 
  // Remember, the array is twice the number of notes (notes + durations)
  for (int thisNote = 0; thisNote < notes * 2; thisNote = thisNote + 2) {

    // calculates the duration of each note
    divider = melody[thisNote + 1];
    if (divider > 0) {
      // regular note, just proceed
      noteDuration = (wholenote) / divider;
    } else if (divider < 0) {
      // dotted notes are represented with negative durations!!
      noteDuration = (wholenote) / abs(divider);
      noteDuration *= 1.5; // increases the duration in half for dotted notes
    }

    // we only play the note for 90% of the duration, leaving 10% as a pause
    tone(buzzer, melody[thisNote], noteDuration*0.9);

    // Wait for the specief duration before playing the next note.
    delay(noteDuration);
    
    // stop the waveform generation before the next note.
    noTone(buzzer);
  }
}`

robsoncouto, 2022-11-12:
Hi Pedro,

That is happening because your music function (as the code in this repo) is blocking code. If you want to know more about the subject, try looking up "blocking vs non-blocking code". The Music() function will prevent the execution of the rest of the code (including reading buttons) until it is finished. So what you want to do is to let the music play "on the background" while alternating with other functions. 

I already did that with my tetris on the gm328a project, so it should be an easy fix.

First have a look at the code  for playing the nodes [here](https://github.com/robsoncouto/gm328a_rev_eng/blob/master/tetris/sound.cpp)

Then you just need to call the sound updating function every cycle like this:

```
void loop() {
  soundTick(); //You can just change Music() to look like mine and call it instead.
  //everything else
}
```
STILL, you don't want to use delay on the loop funcion. Since that will also block the rest of the code(even without Music() you take 2 seconds to read buttons again, that's "0.5Hz"). So you may want to use millis instead in your functions as well. The example [BlinkWithouDelay](https://docs.arduino.cc/built-in-examples/digital/BlinkWithoutDelay) may be helpful. Besides, why are you blinking the buzzer?? Anyways, Good luck!

Grande abraço!
Rob

closed 2022-11-13 by robsoncouto

PedroVillasBoas, 2022-11-14:
Rob,
 First of all, thank you very much for the reply and sorry for the delay in giving a response!
 I didn't understand Blinking buzzer, I just started with Arduino and I'm learning little by little! (If you could talk a little about the problem, I would be very happy!!)
 I haven't made the suggestions you gave yet, but I will make them and come back here with an answer!
 Again, thank you very much! I spent some time trying to decipher what was going on, but you gave me a path to follow!

////////////////////////////////////////////// Translation to portuguese bellow ! //////////////////////////////////////////////

Rob, 
 Primeiramente muito obrigado pelo resposta e perdão pela demora pra dar alguma posição! 
 Não entendi o Blinking buzzer, comecei a pouco com o arduino e estou aprendendo aos poucos! (Se puder dissertar um pouco sobre o problema, ficaria bem feliz!!)
 Ainda não realizei as sugestões que você deu, mas irei realiza-las e volto aqui com uma resposta! 
 Novamente, muito obrigado! Passei algum tempo tentando decifrar o que estava acontecendo, mas você me deu uma grande luz!

robsoncouto, 2022-11-18:
Oi Pedro

Não reparei no comentário devido a formatação do github hehe . O que eu chamei de blinking é ligar e desligar o pino do buzzer. Provavelemente seria melhor apagar de vez isso do seu código, deixando assim:

```
void loop() {
  Music();
  ButtonPressing();
}
```